sparkfun/SparkFun_VL53L5CX_Arduino_Library #2: Two missing API calls limits usefulness of range driven interrupts

chipmc, 2022-01-03:
The Sparkfun Library is missing two functions that that are part of the ST API.

https://www.st.com/resource/en/datasheet/vl53l1x.pdf

In the ST user manual on page 13, these are the parameters to set for range threshold:
Example:
Detectionconfig.DetectionMode = 1
Detectionconfig.Distance.CrossMode = 3
Detectionconfig.IntrNoTarget = 0
Detectionconfig.Distance.High = 1000
Detectionconfig.Distance.Low = 100
Status = VL53L1_SetThresholdConfig(&VL53L1Dev, &detectionConfig );
This function is used to program the device to report ranging only when an object is detected within 10 cm and 1 m (as in this example).
The function VL53L1_GetThresholdConfig() allows the programmed report threshold configuration to be obatined.

There are two items in this configuration that are missing in the SparkFun library : IntrNoTarget and DetectionMode 

Without these two, there are a large number of "false" interrupts where the interrupt is raised but there is not a measurement in the designed range. 

Adding these two API calls is not straight forward as ST does not expose the registers. 

Would it be possible to add these two so a range limited interrupt implementation is possible?

PPVJM, 2022-01-06:
Hi, I will check on that and will get back to you as soon as I have anything new.

PPVJM, 2022-01-10:
This library is for VL53L**5C**X and not VL53L1X. Can you check if you're using the correct library for your part?

chipmc, 2022-01-10:
Sorry, you are correct, I put this on the wrong library.  I have added my request to the VL53L1X library as an issue - unfortunately, there are other similar requests and they have been open for up to 5 months.  I guess Sparkfun may have moved on even though this is an active part.

closed 2022-01-10 by chipmc

PPVJM, 2022-01-10:
> Sorry, you are correct, I put this on the wrong library. I have added my request to the VL53L1X library as an issue - unfortunately, there are other similar requests and they have been open for up to 5 months. I guess Sparkfun may have moved on even though this is an active part.

I will pass this along to the team. Thanks for your support!