sparkfun/SparkFun_VL53L5CX_Arduino_Library #4: Examples fail to run on ESP8266

jarkman, 2022-02-10:


PPVJM, 2022-02-14:
I was able to reproduce the issue and found the cause of the problem. Will work on a fix on the next days.

jarkman, 2022-02-14:
Thanks! There's also some helpful-looking comment in the forum, I haven't tried the suggested workaround yet:
https://forum.sparkfun.com/viewtopic.php?f=68&t=57066&sid=ff28e9a5a183a7748b1d82aefe5363cf

nseidle, 2022-03-03:
This was resolved [via forums](https://forum.sparkfun.com/viewtopic.php?f=68&t=57066&sid=ff28e9a5a183a7748b1d82aefe5363cf:

"It's now working if you set Arduino IDE -> Tools -> "Byte/Word access to IRAM/PROGME (very slow)"
In addition, disable the software watchdog with ESP.wdtDisable() before if (myImager.begin() == false) and then re-enable it with ESP.wdtEnable(3000)"

closed 2022-03-03 by nseidle