sparkfun/SparkFun_VL53L5CX_Arduino_Library #5: Example9_TwoSensors doesn't work

Dave-Da, 2022-03-03:
Trying to run Example9_TwoSensors on an ESP32.

First, i always got this error: ""Sensor 2 not found. Check wiring. Freezing..."

In the example code, you are holding the second sensor in reset by pulling the RST pin to LOW and releasing it from reset by pulling the RST pin to HIGH, which is wrong. According to ST's datasheet it has to be the other way round:

"I2C interface reset pin, active high. Toggle this
pin from 0 to 1, then back to 0 to reset the I2C
slave. Connect to GND via 47 kΩ resistor."

After fixing this, it still doesnt work. I now can connect to both sensors and get the data from the first sensor. But after the data from the first sensor is read, the isDataReady() function for the second sensor always returns FALSE.

I would guess it's something in the library, but couldn't find it yet. 

I would appreciate your help. Thank you

nseidle, 2022-03-03:
Good catch. Fixed in v1.0.1.

[image]

With your fix in place I'm able to get two sensors talking nicely. Be sure to power cycle the sensors before running the example code (pull the Qwiic cable and re-insert) because sensor 1 will have its address correctly set to 0x44, and if you just reset you ESP32 it will fail to connect to sensor 1 because it expects it at the factory default address of 0x29. Alternatively, hook both unit's reset pins up and modify the code so that you are resetting both units after every power cycle.

nseidle, 2022-03-03:
Hmm - lemme just modify the code to do that... hold on.

nseidle, 2022-03-03:
Hmm - it's more broken than I thought. Thanks for reporting. I'll see if I can get it fixed.

Dave-Da, 2022-03-04:
Wow, thank you very much for the fast fix. Works now for me with 2 sensors.

closed 2022-03-04 by Dave-Da